iafilatov/libfprint #2: add touch pad from Xiaomi Mi Notebook Pro

mikey0000, 2017-12-10:
04f3:0c1a is the device ID, I am happy to compile etc to test. Even write code once pointed in the right direction. Thanks.

iafilatov, 2018-01-09:
@mikey0000 For now you can refer to frpint mailing list. There's a number of threads which could help
https://lists.freedesktop.org/archives/fprint/2017-March/000868.html
https://lists.freedesktop.org/archives/fprint/2017-November/000893.html
https://lists.freedesktop.org/archives/fprint/2017-November/000895.html
https://lists.freedesktop.org/archives/fprint/2018-January/000920.html

[This commit](https://cgit.freedesktop.org/libfprint/libfprint/commit/?id=cfe60c0640e445d4bf0af25482f5187a89bc643b) contains an entire set of changes needed to add a completely new driver. If there's already one for a similar device, it should require even less changes.

In general, you'd need to
1. Reverse-engineer the protocol that your device uses to start the device, capture an image of the finger and deactivate.
2. Code up a driver that would send the commands. libfprint in general doesn't care what the driver does. It calls the 'activate' subroutine and waits for the driver to submit the image. It can also call 'deactivate' at any point and the driver should handle it gracefully.
3. Check that everything works in `libfprint/examples`, [fprint_demo](https://www.freedesktop.org/wiki/Software/fprint/fprint_demo/) and [frpintd](https://www.freedesktop.org/wiki/Software/fprint/fprintd/) (the daemon which handles fingerprint auth).

mikey0000, 2018-01-09:
Thanks I'll take a look soon.

mikey0000, 2018-01-17:
on a side note, testing with fingerprint gui, it recognised the swipe when I selected the unknown device (elan) so the existing driver should be a good starting point.

iafilatov, 2018-01-17:
Thanks. We now have a lot more info on the protocol and modes of these readers today (see the mailing list). Hopefully I'll be able to integrate it into the driver soon.

Garuda44, 2018-01-18:
I also have request for the support of fingerprint sensor at Xiaomi Mi Notebook. If I could help somehow I'll be happy to do it, but I cannot do reverse engineering.

jknvv13, 2018-01-30:
Ok, Mi Notebook Pro here.
Actually it... kinda works. But it requests to swipe the finger instead of tapping, fprint says:
Enroll result: enroll-swipe-too-short ←← If you lift the finger fast.
Enroll result: enroll-retry-scan ←← If you do not swipe it slowly, no matter how much you left the finger on the sensor, it requires to swipe.

jknvv13, 2018-01-30:
Oh, and actually "img_capture" gets this, I don't know if this is normal or what.

[finger.zip](https://github.com/iafilatov/libfprint/files/1677913/finger.zip)

iafilatov, 2018-01-30:
@jknvv13 Well... there's a reason it wants you to swipe. Can't make these elans work otherwise. Please see https://bugs.launchpad.net/ubuntu/+source/libfprint/+bug/1641290/comments/41

jknvv13, 2018-01-30:
This is a fprint issue or something on the driver actually?

iafilatov, 2018-01-30:
This is because of the algo that libfprint uses for matching. It's common for all drivers. That algo works best when it has at least 60% of the fingerprint. Compare that to the area that your reader can cover. In my case it's 20-25%.

jknvv13, 2018-01-30:
And can this be solved or this will be stuck forever in a "swipe" world?
🤦‍♂️

iafilatov, 2018-01-30:
It can if someone implements a different algo or finds a usable lib that can be distributed with libfprint. Not a simple task. I wouldn't say "forever" but for now it's gonna be swipe.

Garuda44, 2018-01-31:
Sorry but my Xiaomi doesn't see the device.

iafilatov, 2018-02-22:
@Garuda44 I've added more devices to the supported list. Could you please try the latest code?

Dreamwork57, 2018-02-24:
First, I wanna thank you all for trying to bring elan fingerprint readers to work.

I've tried out this driver for my Xiaomi Mi Notebook Pro, because I needed a driver that worked with the fingerprint reader.
Right now it does work, meaning it recognize the hardware, and my finger (that's the good part).
Now, beyond the fact you must swipe to use the sensor, it also cannot log me. And of the top of that, if I swipe too quickly, I've got the error mentioned by @jknvv13 : 

> Enroll result: enroll-swipe-too-short ←← If you lift the finger fast.

The problem is that when I got this error, the fprint doesn't react anymore, it just kind of freezes and so I cannot log.
Also another bug (wich I think is more because of sddm) : everything I described until now is when I log just after closing the session. But when I restart my computer, I cannot use fingerprint to log AND the password doesn't work at all. But I think it's really a bug in sddm.

iafilatov, 2018-02-25:
@Dreamwork57 Can you please post the logs from examples/enroll and examples/verify? There's also the last enrolled image from examples/enroll called enrolled.pgm. If you don't mind, could you please send it to me ( my github handle at gmail )? If you do, please just describe what you see. Assembling of frames should work correctly with the latest code but from what people report, I still have doubts.

The various XYZDM's indeed lack a good integration with fingerprint devices. If the reader works fine, you normally don't get any problems. But if you need to fall back to password, yeah that's not always easy. Besides, passwords are still required for things like encrypted homes, keyrings, so...

IsoaSFlus, 2018-02-26:
@Dreamwork57  I got the same problem. And I'm MiNote 13.3. It always get  "swipe too short" and doesn't respond any more. This problem makes it hard to enroll a fingerprint. Although I have enrolled my fingerprint successfully five times, I cannot verify my fingerprint correctly.

iafilatov, 2018-02-26:
>  It always get "swipe too short" and doesn't respond any more.

Ok, found it. Should be fixed by ea406e4.

mikey0000, 2018-02-26:
Awesome that should fix my issues too.

Dreamwork57, 2018-02-26:
Thanks for the answer and the fix, I'll test it as soon as released.
It would be great to make this sensor work.

iafilatov, 2018-02-26:
This won't get to the upstream for a while. The only way to use it (except with 0907) for now is to clone this repo, build and install.

IsoaSFlus, 2018-02-27:
@iafilatov  Thanks for your effort. After some tests, I think it has fixed freezing on enroll with swipe too short. But on verifying,  when I get "swipe too short", it will be freezed without any retry.

IsoaSFlus, 2018-02-28:
@iafilatov GREAT! It works perfect. Thank you very much!

iafilatov, 2018-02-28:
@mikey0000 @Dreamwork57 Please see if you still have the lock-ups, if not, I'll close this issue.

Dreamwork57, 2018-02-28:
@iafilatov I won't be able to test it out until this very afternoon (in 8-9hours, don't know were you're located)
So i'll tell you then

mikey0000, 2018-02-28:
I'll give it a go

Dreamwork57, 2018-02-28:
Ok so now it actually works, the enroll is working perfectly but I still have some problems to login with the fingerprint. Pretty sure that it doesn't have anything to do with the driver, but if someone could give me an answer, I'll be thankful : 
When I lock the computer and then try to login, at first I can't use the fingerprint (it doesn't work, doesn't make any life sign) so I tap my password and it asks me to log with fingerprint. I do it, it works and I must again hit "Enter" to be logged in. Maybe it as to do with my pam conf I don't know. If someone can help me contact me in private. I use debian 9 stretch by the way.

mikey0000, 2018-02-28:
the fprint-demo app doesn't seem to work now. Dunno whats going on there, however the fingerprint-gui works great. Scanning is definitely working well as well as verifying. Will check login.

mikey0000, 2018-02-28:
having udev issues trying to get fingerprint-gui to work without sudo.. getting there.

closed 2018-02-28 by mikey0000

kjhoerr, 2018-03-11:
Not sure if a different Arch group exists for managing devices, but I added the `plugdev` group as @mikey0000 mentioned. In Arch you can't just change group/ownership of a device though, since that gets reset when rebooting. I had to create a rules file for udev:

    # /etc/udev/rules.d/elan-fprint.rules
    ATTR{idVendor}=="04f3", ATTR{idProduct}=="0c1a", GROUP="plugdev"

Then all that was required then was making sure that udev could read the new rules:

    # udevadm trigger

And it started working automatically!

arogozhnikov, 2018-05-31:
Hey, as some kind of feedback - I've successfully installed the latest version, but verification results in too many false negatives (1 successful in ~10 attempts)

vasiliy-garenskikh, 2018-07-21:
Hi everyone, can u make full instruction?
I can't activate fingerprint on my Xiaomi mi notebook pro with Manjaro KDE, because Fingerprint GUI app doesn't see fingerprint module..

LordJABA, 2018-07-27:
be sure to uninstall/purge the lib installed from system repos, I had the same  and this solved the issue. If you want I can give you trimmed bash_history dump on my successful attempt on kubuntu 18.04

kovalenp, 2018-12-10:
@LordJABA yes please, I wonder how I can enable fingerprint reader in 18.10

Drun555, 2019-01-28:
04f3 in Xiaomi Notebook Pro is completely unusable. Is here any moves in last months?

andirsun, 2020-04-26:
> 04f3 in Xiaomi Notebook Pro is completely unusable. Is here any moves in last months?

Bro I have The same laptop, how can I activate my fingerprint with fingerprint gui ? im in elementary os