iafilatov/libfprint #7: Scanner rotated on Acer Swift 5

maciel310, 2018-02-27:
It appears that on the Acer Swift 5 the fingerprint sensor is rotated 90 degrees from where this driver expects it to be, resulting in images that don't stitch together properly. Seems like the code assumes that the driver is positioned like this:

XXXXXXX
XXXXXXX
XXXXXXX

But on my laptop it is like this:

XXXX
XXXX
XXXX
XXXX

The code works otherwise for scanning the print, it just stitches the image together incorrectly. If I swipe left to right instead of top to bottom then it produces a good image. (For some reason verification fails still, but it is at least giving a good image that way. I can open a separate issue for the verify failure.)

Not sure how the stitching works currently, but it seem to do some sort of overlap detection to help with the alignment. If there's a way of quantifying how well the images overlap maybe run the process in both orientations, that way it will work either way that you scan your print?

By the way, thanks a ton for working on this driver!

iafilatov, 2018-02-27:
Alas, one of my greatest fears comes true! The rotated scanner. The problem is, libfprint already tries to stitch in two directions: bottom to top and top to bottom (it then selects the best image). To accommodate 90 deg rotated readers it would need to rotate every frame and repeat the procedure, resulting in 4 assembly attempts overall. I don't think the maintainers will like it. Unless it's a special case just for elan, but still. I guess I'll need to ask them.

What's your device id? Maybe we're lucky and only certain devices are installed rotated.

maciel310, 2018-02-27:
Pretty tame for your greatest fear ;)

The device ID from lsusb is 04f3:0c03

iafilatov, 2018-02-27:
I only have one report of 0c03 and I don't know if it's rotated (in fact, it's NOT rotated, all other are). So I guess I can just special-case your device and see what happens.

maciel310, 2018-02-27:
Sounds good. Let me know when there's something to test and I'll pull and give it a go. Thanks for looking!

iafilatov, 2018-02-28:
I've made some changes, please see it they help.