iafilatov/libfprint #10: No prerequisite list found

stefanocirici, 2018-05-04:
I had problems compiling cause many libs were missing.
Here it is a list of what I had to install to make it working: 
- `pkg-config`
- `libtool`
- `m4`
- `automake`
- `libusb-1.0.0-dev` (I had to compile it from [source](https://github.com/libusb/libusb/releases))
- `libudev-dev`
- `libnss3-dev`
- `libglib2.0-dev`
- `libpixman-1-dev`
- `libxv-dev`

Probably the list it's not complete but I am quite sure that all of these packages are needed.
It would be helpful adding a complete list in the README.md file.

FliiFe, 2018-05-10:
I did not have to compile libusb from source, I just installed the package (also, I have both `-dev` and non-`dev` variants, but I'm guessing only the `dev` variant is used)

iafilatov, 2018-06-24:
https://github.com/iafilatov/libfprint#pre-requisites

closed 2018-06-24 by iafilatov