iafilatov/libfprint #12: Trying to make 04F3:3057 work

FliiFe, 2018-05-10:
I have an `ELAN1300:00  04F3:3057` touchpad with included fingerprint sensor. The thing is, I'm not even sure the fingerprint sensor should be listed separately as its own input device, but it's currently not (in `/proc/bus/input/devices` only the touchpad is listed).

Then I'd like to see if this driver can work out of the box for this fingerprint sensor (I don't know any C at all, unfortunately)

iafilatov, 2018-05-15:
So far all Elan fp sensors recognized by this driver have come up as separate devices (i.e. in lsusb), even integrated in touchpad. So normally I'd expect it to show. Unless it's some kind of new device that's sooo integrated it needs just one driver. If you want try attaching this driver, copy the line at https://github.com/iafilatov/libfprint/blob/elan/libfprint/drivers/elan.h#L158 and set `.product = 0x3057`. Then compile and try it out like in readme. But I wouldn't expect much. It could well be using a completely different protocol.

iafilatov, 2018-06-02:
There's now a how-to [here](https://github.com/iafilatov/libfprint#unrecognized-devices).

closed 2018-06-09 by iafilatov

Man-Jain, 2020-05-28:
Hey Can you confirm if this worked for you ? @FliiFe

FliiFe, 2020-05-28:
It has not, I have yet to get this sensor to work @Man-Jain

Man-Jain, 2020-05-28:
Oh! Thanks

Leatherface75, 2020-05-31:
Same problem here and it doesn't even show in lsusb

`Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 13d3:3526 IMC Networks 
Bus 001 Device 003: ID 13d3:5a07 IMC Networks 
Bus 001 Device 002: ID 046d:c01e Logitech, Inc. MX518 Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

mincrmatt12, 2020-08-10:
I have a prototype driver for these sensors on my fork of libfprint at [mincrmatt12/libfprint](https://github.com/mincrmatt12/libfprint/tree/mincrmatt12/elan-spi), although because these sensors talk via an amalgamation of SPI and vendor-defined HID commands it requires a bit of system setup to get working (see the README [here](https://github.com/mincrmatt12/elan-spi-fingerprint)). It works for the one system I've tested it with, but YMMV. It also has the same problem that all the sensors here do, which is that it's way too small for the normal libfprint matching algorithm, so you have to swipe your finger across the sensor (which is a bit annoying given that it's embedded in the touchpad)