iafilatov/libfprint #13: Support for 2542

jpreston84, 2018-06-02:
Hi,

I have an Asus Vivobook 14 Flip. It has an Elan fingerprint reader, which shows up as 04f3:2542. I was wondering how hard it would be to add support for this device to your driver.

Here's the output of lsusb...

Bus 002 Device 009: ID 0bda:8153 Realtek Semiconductor Corp. 
Bus 002 Device 008: ID 2109:0813 VIA Labs, Inc. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 8087:0a2b Intel Corp. 
Bus 001 Device 004: ID 04f3:2542 Elan Microelectronics Corp. 
Bus 001 Device 003: ID 13d3:5a03 IMC Networks 
Bus 001 Device 002: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 009: ID 2109:2813 VIA Labs, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Is there anything I can do to facilitate adding support for this device?

iafilatov, 2018-06-02:
Hi,

I wrote up a [how-to on adding new devices to the driver](https://github.com/iafilatov/libfprint#unrecognized-devices). If your reader responds in any way, then it could work. If not, it's probably an entirely different device with a different protocol, and this driver is useless for it.

jpreston84, 2018-06-02:
All right -- I followed your instructions (after installing a large number of dev packages). Here's what I got...

(venv) root@MyComputer:~/libfprint/builddir_dbg# examples/img_capture
Found device claimed by ElanTech Fingerprint Sensor driver
elan:error [dev_init] could not claim interface 0: LIBUSB_ERROR_BUSY
async:error [fp_async_dev_open] device initialisation failed, driver=elan
Could not open device.
(venv) root@MyComputer:~/libfprint/builddir_dbg# 

Any thoughts on what to do next?

iafilatov, 2018-06-03:
"Device busy" means it's already open by another driver. So it's not an fp reader. Maybe a touchpad. Or maybe scanner and touchpad register as a single device. Unfortunately, it's not possible to talk to such readers with libfprint.

closed 2018-06-09 by iafilatov