iafilatov/libfprint #14: Native dependency "udev" not found, but libgudev & libgudev-devel installed

moore-bryan, 2018-06-26:
Any thoughts?
[meson-log.txt](https://github.com/iafilatov/libfprint/files/2137606/meson-log.txt)

iafilatov, 2018-06-26:
libudev-devel?

moore-bryan, 2018-06-26:
I'm on Solus and that's the "dev" name...

iafilatov, 2018-06-26:
Without the "g". libudev and libgudev are different projects. You need headers for libudev.

moore-bryan, 2018-06-26:
Solus doesn't have libudev, only libgudev; well, it does have python3-pyudev, but I already have that installed and still no love.

Guess I'll need to go compiling!

moore-bryan, 2018-06-26:
It looks as though installing systemd-devel on Solus fixes this issue... thanks for being my sounding board!

closed 2018-06-26 by moore-bryan

iafilatov, 2018-06-26:
Here's what I found https://solus-project.com/forums/viewtopic.php?t=10141

qsantos, 2025-03-12:
Same issue on Debian, had to install `systemd-dev`