iafilatov/libfprint #15: Tried to access unknown option "drivers"

mperochon, 2018-07-17:
Hello,

when i tried to compile the library, i have this error : 
`Called `/usr/bin/pkg-config libusb-1.0 --libs` -> 0
-lusb-1.0
Native dependency libusb-1.0 found: YES 1.0.21
Running compile:
Working directory:  /tmp/tmp_qskrjw6
Command line:  cc /tmp/tmp_qskrjw6/testfile.c -pipe -D_FILE_OFFSET_BITS=64 -o /tmp/tmp_qskrjw6/output.exe -O0 -Wl,--start-group -lm -Wl,--end-group 

Code:
 int main(int argc, char **argv) { return 0; }
Compiler stdout:
 
Compiler stderr:
 
Library m found: YES

meson.build:46:0: ERROR:  Tried to access unknown option "drivers".`

I use ubuntu 18.04 and the kernel 4.17.6-041706-generic.

Thanks a lot :)

iafilatov, 2018-07-17:
Could you please post the entire output of `meson builddir`? And of `meson configure builddir` as well.

iafilatov, 2018-11-09:
Please reopen if you still have compile errors.

closed 2018-11-09 by iafilatov