iafilatov/libfprint #17: Not device found

miroslav-lee, 2018-07-22:
I run command LD_LIBRARY_PATH=./libfprint/.libs/ fprint_demo and see message "Device not found" ((
[image]

iafilatov, 2018-07-22:
`LD_LIBRARY_PATH=./libfprint/.libs/` looks like what you had to set when building with autotools. If you build with meson in builddir, you need to run `LD_LIBRARY_PATH=libfprint fprint_demo` (note to self: update readme).

miroslav-lee, 2018-07-23:
Thanks, it`s works!))
But i don`t now, how setting login from fprint_demo or fingerprint gui?

iafilatov, 2018-07-23:
fprint_demo is basically just that - for demo purposes. Not for login etc. Not sure about fingerprint gui, never used it. I think earlier these progs used to share print data with system fprintd daemon but that has changed. You need fprintd and libpam-fprint for login. Once you have them, run fprintd-enroll. But first, make sure you can verify comfortably using i.e. fprint_demo. Otherwise logging in and/or running sudo will be pain.

miroslav-lee, 2018-07-23:
[image]

It`s ok?

iafilatov, 2018-07-23:
fprintd-enroll is a tool that communicates with fprintd daemon, which in turn needs to be linked to the correct version of libfprint. So you shouldn't need LD_LIBRARY_PATH for fprintd-enroll. Instead, you need to install libfrint from this repo system-wide (`sudo ninja install`) and restart fprintd (depends on your distro, try `sudo service fprintd restart`). If it still doesn't work try `sudo ldconfig`, then restart.

The image looks quite good. Your reader should be usable, esp. once you get the hang of swiping consistently (took me some time, ymmv).

closed 2018-08-10 by iafilatov