iafilatov/libfprint #18: Problem with fprintd-enroll

gish94, 2018-08-13:
I have done all things as in instructions and i can get my fingerprint image from
sudo ./examples/img_capture 
but after ninja install, doing "apt install libpam-fprintd" and making fprintd-enroll i am getting "list_devices failed: No devices available" 
what i am doing wrong? 

here is my Elan device log from lsusb
Bus 001 Device 003: ID 04f3:0c1a Elan Microelectronics Corp.

iafilatov, 2018-08-13:
Right after you you install all the required packages and libfprint you may need to
1. restart frpintd to link against the correct version of libfprint: `sudo service fprintd restart`
2. if the above doesn't work, check that linker picks the correct version:
`ldd /usr/lib/fprintd/fprintd | grep libfprint`
It should show something like
`libfprint.so.0 => /usr/local/lib/libfprint.so.0 (0x00007fae2d7ed000)`
where the most important part is *local*. There will be another version of libprint installed on your system as dependency of fprintd package which doesn't support your device (spoiler: this is going to change pretty soon). If it's something like /usr/lib/libfprint.so.0, its the wrong one. Try running `sudo ldconfig` and repeating 1, 2.
3. _If_ 1 and 2 don't work you may need to check the configuration of ldconfig to make sure that libs you install from source take precedence over libs from packages. But this should already be the default.

iafilatov, 2018-08-21:
Please reopen if the problem persists.

closed 2018-08-21 by iafilatov