iafilatov/libfprint #21: Unrecognized error code when running img_capture with Elan 04f3:0903

neslinesli93, 2018-08-28:
Hi,
thanks for the project - awesome stuff!

I have been trying to setup the whole libfprint/fprintd stuff today on my zenbook ux430.
I've tried to run your examples as well, and while `verify` and `enroll` seem to work perfectly, I had a strange issue when running `img_capture`.

I am pasting the logs here:
```
Found device claimed by ElanTech Fingerprint Sensor driver
Opened device. It's now time to scan your finger.

sync:error [fp_dev_img_capture] unrecognised return code 101
image capture failed, code -22
```

It looks like `fp_dev_img_capture` is getting this weird 101 error code when calling `fp_async_capture_start`... maybe it's a driver issue?

Thanks again!

iafilatov, 2018-08-28:
Hi,

Could you please post the entire output of `img_capture`?

neslinesli93, 2018-08-28:
That is the entire output... should I try compiling with the debug flag to see if there are more logs?

iafilatov, 2018-08-28:
Oh I see.. please try elan-glib-2.50 branch, or better yet, master from https://gitlab.freedesktop.org/libfprint/libfprint

neslinesli93, 2018-08-28:
Okay, I used the libfprint master, and this is the log `img_capture` produced when I quickly put my finger on the sensor and then remove it: https://pastebin.com/dTbpKgcv

iafilatov, 2018-08-28:
You need to swipe. The driver aborts because there's not enough frames (line 204).

iafilatov, 2018-08-28:
There's a [note](https://github.com/iafilatov/libfprint#common-problems) in readme that describes the problem. It's not about resolution or number of images per se, it's just about the "area" of the print that's  available for processing. The whole point of swiping is to expand that area and capture more minutiae.

The algo you link to is also based on minutiae, so I assume it will have the same problems with small cut-out prints as NBIS (which btw will updated to a recent version by libfrpint maintainer in the next release).

Last time I researched this topic there were no open source implementation of fingerprint algos suitable for small sensors like the ones in smartphones and recent laptops. Under Windows/Android/etc., I believe, an altogether different approach is used. It could be based on something other that minutiae or capture some additional properties. If someone actually implemented something like that under a suitable license, it would be a godsend. There's gonna be more and more tiny touch readers in the future and libfprint risks falling behind the curve. When I started this project, I only had a usb reader I bought on ali express. I had no idea they were putting very similar Elan's in ASUS and Xiaomi and that many people would actually need this driver...

If you wish to discuss replacements for NBIS or have other ideas to share, please consider starting a thread on the mailing list so that more people can benefit from the discussion ([although you wouldn't be the first one to do so...](https://lists.freedesktop.org/archives/fprint/2018-January/000966.html) and we still don't have a solution).

closed 2018-11-09 by iafilatov