iafilatov/libfprint #22: After run command 'ninja', i noticed error

miroslav-lee, 2018-10-28:
I run commands:
```
venv/bin/meson builddir_dbg
venv/bin/meson configure builddir_dbg -Ddebug_log=true
cd builddir_dbg
ninja
```
But after `ninja`, I noticed error: `../libfprint/drivers/elan.c:827:2: error: implicit declaration of function ‘G_DEBUG_HERE’ [-Werror=implicit-function-declaration] `

iafilatov, 2018-10-28:
First of all, I'd recommend you to try building `elan` (or the default `elan-github-msg`, they're the same) branch. If you tried that it should've failed but I've just pushed a fix so can you please try again?
As for updating glib, the best thing to do is to upgrade your linux distro to a newer version... glib is a core package so I really don't recommend trying to install a version that's not distributed through the original package channel. Things might break badly. For example, speaking ubuntu, you need bionic (18.04) or newer.

miroslav-lee, 2018-10-28:
Is your [manual](https://github.com/iafilatov/libfprint/blob/elan-github-msg/README.md) actual?

iafilatov, 2018-10-29:
It is, although if your package manager contains a recent version of meson, building can be even simpler, e.g.
```
apt install ninja-build meson
meson builddir_dbg
meson configure builddir_dbg -Ddebug_log=true
cd builddir_dbg
ninja
```
The problems section is, unfortunately, still valid.

closed 2019-05-12 by iafilatov