iafilatov/libfprint #25: Asus UX461UN | Unrecognized device

andrzejpakula, 2019-01-13:
Hey. I successfully installed your program on ubuntu 18.10. Unfortunately, my device is unrecognized on my new Asus UX461UN. I also realized that the lsusb | command grep -i elan does not show my fingerprint reader device info. Here is the thread about it https://ubuntuforums.org/showthread.php?t=2388793 . Unfortunately, I did not find any more information. I can not find any driver. What can you do about this? Is there a chance that the driver will appear? Best Regards

iafilatov, 2019-01-13:
Hi Andrzej, it looks like your fingerprint sensor is connected to SPI bus. libfprint only supports USB devices at the moment so unfortunately this driver won't work for you. Sorry..

andrzejpakula, 2019-01-13:
Ok, it's hard. Thanks for your response.

closed 2019-01-13 by andrzejpakula