iafilatov/libfprint #27: I own a 04f3:0903 (Asus Zenbook 430) and willing to hack

fransschreuder, 2019-05-10:
Hi, 
I just bought this new Zenbook with the 04f3:090 fingerprint reader and with the 0.99 version of libfprint the fingerprint reader works to some extent, it sometimes successfully enrolls, but most of the time exits with: enroll-unknown-error.

I can program / hack / test, just need some pointers on where to start.
I don't know whether I better start here, or with the official libfprint repo, I guess it is good to use the official code since it was merged there, but maybe this is the place for good pointers to the specific hardware.

iafilatov, 2019-05-12:
Hey Frans,

Your symptoms sound like [small sensor problem](https://github.com/iafilatov/libfprint#common-problems). Or, less likely, a [rorated sensor](https://github.com/iafilatov/libfprint/issues/23). See the log snippet to find out how many minutiae libfprint is able to detect (you probably want to pipe and grep "minutiae" - the log has become a lot less readable in recent releases). If it constantly gets 10 or less, there's the problem. Or check the images visually. If they don't immediately resemble fingerprints, it's the same thing. 

If you want to hack, I'd recommend going for the official repo. The code is much newer there. But if I'm right, hacking won't help as long as no dedicated algo exists in libfprint.

fransschreuder, 2019-05-12:
Thanks for the pointers. I will have a look into it. 
I can actually log in now with the fingerprint sensor, but it is a bit unreliable, I often have to try a couple of times.

closed 2019-05-28 by iafilatov