iafilatov/libfprint #29: Lockout after cancelling enroll or verify operation

nieltg, 2019-06-13:
I own `04f3:0903` too. Thanks for this ELAN driver. Now I can log in with the fingerprint. I learned how to get a good fingerprint scan by using [gtk-libfprint-test](https://gitlab.freedesktop.org/libfprint/libfprint/tree/66891274a7fbfb7b0c97c271aebad87a083bc612/demo) and match it with the sample images provided in this repository.

Here is what I learned:
- Swype down, not swype up.
- Touch gently. Don't have to touch too hard or too light.

I noticed that the fingerprint device will locked up when I cancelled enrolling or verifying. Here is how it looks like:

```
➜  ~/Projects/libfprint/libfprint git:(1abe213) ✗ fprintd-verify
Using device /net/reactivated/Fprint/Device/0
Listing enrolled fingers:
 - #0: right-index-finger
^C
➜  ~/Projects/libfprint/libfprint git:(1abe213) ✗ fprintd-verify
Using device /net/reactivated/Fprint/Device/0
failed to claim device: Device was already claimed
```

I have to reboot my laptop so the fingerprint can be unlocked. I'm willing to help, but I'm still beginner here. Just tell me how to do it if you need something, like logs, etc. Thank you very much.

iafilatov, 2019-06-13:
Hey Daniel, good to hear that it works for you, especially 0903. The direction of swipe used to matter a while ago when there was a bug in the assembling code and one direction worked better than the other. But it could be just that it's easier for you to consistently swipe down than up on your laptop. Which is ok too.

I'm away from my device at the moment so I cant reproduce this issue but I'm guessing that briefly touching your scanner might unlock without rebooting. I'll take a closer look later.

iafilatov, 2019-07-14:
I found the change upstream that brought this bug but apparently it was fixing other bugs, so I'm not realy sure how to solve this. For now one thing I can say is, this has never been an issue for me when using fingerprint for login, sudo etc. fprintd daemon never seems to cancel the process the way it happens when you Ctrl+C the `frpintd-*` commands.

nieltg, 2019-07-15:
Okay, than I should report this to the upstream repository.

Thank you very much, @iafilatov

closed 2019-07-15 by nieltg