iafilatov/libfprint #31: when i run command sudo examples/verify some error is appar.

handbye, 2019-11-17:
```shell
dk@dk:~/libfprint/builddir_dbg$ sudo examples/verify
fp:debug [fp_init] 
fp:debug [register_driver] registered driver upekts
fp:debug [register_driver] registered driver aes3500
fp:debug [register_driver] registered driver aes4000
fp:debug [register_driver] registered driver aes2501
fp:debug [register_driver] registered driver aes2550
fp:debug [register_driver] registered driver uru4000
fp:debug [register_driver] registered driver vcom5s
fp:debug [register_driver] registered driver upeksonly
fp:debug [register_driver] registered driver aes1610
fp:debug [register_driver] registered driver aes1660
fp:debug [register_driver] registered driver aes2660
fp:debug [register_driver] registered driver vfs101
fp:debug [register_driver] registered driver vfs301
fp:debug [register_driver] registered driver vfs5011
fp:debug [register_driver] registered driver upektc
fp:debug [register_driver] registered driver upektc_img
fp:debug [register_driver] registered driver etes603
fp:debug [register_driver] registered driver vfs0050
fp:debug [register_driver] registered driver elan
fp:debug [find_supporting_driver] driver elan supports USB device 04f3:0c1a
fp:debug [find_supporting_driver] selected driver elan supports USB device 04f3:0c1a
Found device claimed by ElanTech Fingerprint Sensor driver
sync:debug [fp_dev_open] 
async:debug [fp_async_dev_open] 
elan:debug [dev_init] 
async:debug [fpi_drvcb_open_complete] status 0
sync:debug [sync_open_cb] status 0
Opened device. Loading previously enrolled right index finger data...
fp:debug [load_from_file] from /home/dk/.fprint/prints/0015/00000000/7
fp:error [load_from_file] /home/dk/.fprint/prints/0015/00000000/7 load failed: Failed to open file “/home/dk/.fprint/prints/0015/00000000/7”: No such file or directory
Failed to load fingerprint, error -2
Did you remember to enroll your right index finger first?
sync:debug [fp_dev_close] 
elan:debug [dev_deinit] 
elan:debug [elan_dev_reset] 
async:debug [fpi_drvcb_close_complete] 
sync:debug [sync_close_cb] 
fp:debug [fp_exit] 
```

iafilatov, 2019-11-20:
It says it cannot find an image of fingerprint on your disk to compare with (to "verify" against). Did you run examples/enroll?

Also, this repo is really old and I'm not updating it. It's better to try stuff in the upstream libfprint https://gitlab.freedesktop.org/libfprint/libfprint

handbye, 2019-11-21:
I have run examples/enroll  and i will try new repo. if succeeful  i'll give you feedback! thanks!

handbye, 2019-11-26:
problem solved

closed 2019-11-26 by handbye