iafilatov/libfprint #34: libfprint.so.0: cannot open shared object file: No such file or directory

initzx, 2020-05-05:
Hi, I am getting this error when i execute fingerprint-identifier or fingerprint-gui.

I have been able to enroll my fingerprint with fprintd-enroll and verify it with fprintd-verify.

How do I fix this?

I have a thinkpad t490s running on kernel 5.6.8-arch1-1.

iafilatov, 2020-05-20:
Looks like you haven't installed libfprint after building it: `sudo ninja install`. If you did that then maybe the shared lib cache hasn't been updated. The safest thing to do is to restart. If it doesn't help, you need to check whether the install prefix, most likely `/usr/local/lib`, is included in library search path. See `/etc/ld.so.conf` and/or `/etc/ld.so.conf.d/`.

Another thing to try is to set `$LD_LIBRARY_PATH`:
```
cd /home/you
git clone ... 
cd libfprint
meson build
cd build
ninja
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/you/libfprint/build/libfprint
fingerptint-gui
```
(for the current session only)

initzx, 2020-05-30:
Hello, it seems that the problem is related to the filename of the object file. 

I have libfprint-2.so, but not libfprint.so.0

Aliasing libfprint-2.so does not work very well either.