iafilatov/libfprint #36: Making the drivers work on 96x96 sensor

nibirrayy, 2021-01-15:
I have a square sensor but I don't know if it is the 96x96 sensor but what I can tell is that the driver doesn't detect my finger print correctly or even if it does in some rare cases it is not sufficient to authenticate.

Is there is a way to make it work?

devmrfitz, 2021-01-15:
I'm also having the same issue

iafilatov, 2021-01-15:
If the scanner is recognized but enrolling and/or recognizing doesn't work well, then likely nothing can be done https://github.com/iafilatov/libfprint#common-problems

nibirrayy, 2021-01-16:
is there any other algorithm that we can use or can we create an image from another sensor and put it to use?

I just want the sensor to work like it does on windows

iafilatov, 2021-01-17:
None that I know of that's available and fits libfprint's license. You can follow the development in the main repo at https://gitlab.freedesktop.org/libfprint/libfprint. It's quite lively and this problem is known. On windows a proprietary driver with a proprietary algo is used (actually, I think the algo is not even a part of the driver at all).

devmrfitz, 2021-01-18:
Is there any proprietary algo available that I can myself compile with
libfprint?

iafilatov, 2021-01-27:
Again, none that I know of.

devmrfitz, 2021-04-08:
Any updates on this issue?

nibirrayy, 2021-04-09:
@devmrfitz no update so far just stop using the sensor and didn't try anything new

gaetan1903, 2021-12-10:
Same issue