iafilatov/libfprint #39: Support for 04f3:0c4b

UnSstrennen, 2021-06-21:
`ID 04f3:0c4b Elan Microelectronics Corp. ELAN:Fingerprint`

Your solve will be unique, because I found out so many sites&repos(

That ID is mounted on Lenovo ThinkPads.

XelXen, 2021-11-12:
Even I am facing that issue, i am trying other branches

listout, 2021-12-06:
@XelXen  Same, I tried build the driver from source. Not working.

p-patil, 2021-12-08:
I also tried building from source and manually adding this vendor ID to the table. Couldn't get `examples/img_capture` to work, and saw this at the end of the logs:
```
libusb: error [udev_hotplug_event] ignoring udev action change
libusb: error [udev_hotplug_event] ignoring udev action bind
```
I have a ThinkBook 14, which uses this fingerprint sensor.

listout, 2021-12-15:
@UnSstrennen Any clues. @iafilatov says that the devices line could be supported (quote). I would like to work on this if there was any help from the maintainers.

> I have reasons to believe that the entire family of compatible Elan readers should already be recognized without modification, so if you need to do this, the chance is slim. But my info could be outdated.

However the on libprintf's gitlab they mentioned something about reverse engineering being required for support.

jackfido, 2022-06-14:
I have the same issue with this driver too, I have to run example/img_capture

`(venv) user@host:~/Downloads/libfprint-elan-github-msg/builddir_dbg$ examples/img_capture
fp:debug [fp_init] 
fp:debug [register_driver] registered driver upekts
fp:debug [register_driver] registered driver aes3500
fp:debug [register_driver] registered driver aes4000
fp:debug [register_driver] registered driver aes2501
fp:debug [register_driver] registered driver aes2550
fp:debug [register_driver] registered driver uru4000
fp:debug [register_driver] registered driver vcom5s
fp:debug [register_driver] registered driver upeksonly
fp:debug [register_driver] registered driver aes1610
fp:debug [register_driver] registered driver aes1660
fp:debug [register_driver] registered driver aes2660
fp:debug [register_driver] registered driver vfs101
fp:debug [register_driver] registered driver vfs301
fp:debug [register_driver] registered driver vfs5011
fp:debug [register_driver] registered driver upektc
fp:debug [register_driver] registered driver upektc_img
fp:debug [register_driver] registered driver etes603
fp:debug [register_driver] registered driver vfs0050
fp:debug [register_driver] registered driver elan
fp:debug [find_supporting_driver] driver elan supports USB device 04f3:0c4b
fp:debug [find_supporting_driver] selected driver elan supports USB device 04f3:0c4b
Found device claimed by ElanTech Fingerprint Sensor driver
sync:debug [fp_dev_open] 
async:debug [fp_async_dev_open] 
**libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/003/008, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
async:error [fp_async_dev_open] usb_open failed, error -3
Could not open device.
fp:debug [fp_exit]** 
`

then as sudo:

`(venv) user@host:~/Downloads/libfprint-elan-github-msg/builddir_dbg$ sudo examples/img_capture
fp:debug [fp_init] 
fp:debug [register_driver] registered driver upekts
fp:debug [register_driver] registered driver aes3500
fp:debug [register_driver] registered driver aes4000
fp:debug [register_driver] registered driver aes2501
fp:debug [register_driver] registered driver aes2550
fp:debug [register_driver] registered driver uru4000
fp:debug [register_driver] registered driver vcom5s
fp:debug [register_driver] registered driver upeksonly
fp:debug [register_driver] registered driver aes1610
fp:debug [register_driver] registered driver aes1660
fp:debug [register_driver] registered driver aes2660
fp:debug [register_driver] registered driver vfs101
fp:debug [register_driver] registered driver vfs301
fp:debug [register_driver] registered driver vfs5011
fp:debug [register_driver] registered driver upektc
fp:debug [register_driver] registered driver upektc_img
fp:debug [register_driver] registered driver etes603
fp:debug [register_driver] registered driver vfs0050
fp:debug [register_driver] registered driver elan
fp:debug [find_supporting_driver] driver elan supports USB device 04f3:0c4b
fp:debug [find_supporting_driver] selected driver elan supports USB device 04f3:0c4b
Found device claimed by ElanTech Fingerprint Sensor driver
sync:debug [fp_dev_open] 
async:debug [fp_async_dev_open] 
elan:debug [dev_init] 
async:debug [fpi_drvcb_open_complete] status 0
sync:debug [sync_open_cb] status 0
**Opened device. It's now time to scan your finger.**

sync:debug [fp_dev_img_capture] to be handled by elan
async:debug [fp_async_capture_start] 
[... 64 lines cut ...]
poll:debug [get_next_timeout_expiry] next timeout in 0.009989s
poll:debug [get_next_timeout_expiry] next timeout in 0.009979s
poll:debug [get_next_timeout_expiry] next timeout in 0.009976s
poll:debug [get_next_timeout_expiry] next timeout in 0.009971s
poll:debug [get_next_timeout_expiry] next timeout in 0.009967s
poll:debug [get_next_timeout_expiry] first timeout already expired
poll:debug [handle_timeout] 
elan:debug [elan_change_state] changing to 3
elan:debug [elan_stop_capture] 
elan:debug [elan_dev_reset] 
drv:debug [__ssm_call_handler] 0x561384f98750 entering state 0
elan:debug [stop_capture_run_state] 
elan:debug [elan_run_cmd] 000b
drv:debug [fpi_ssm_mark_aborted] error -4 from state 0
drv:debug [fpi_ssm_mark_completed] 0x561384f98750 completed with status -4
elan:debug [stop_capture_complete] 
elan:debug [dev_change_state] 3
poll:debug [fpi_timeout_add] in 10ms
poll:debug [get_next_timeout_expiry] next timeout in 0.009998s
poll:debug [get_next_timeout_expiry] first timeout already expired
poll:debug [handle_timeout] 
elan:debug [elan_change_state] already in 3
libusb: error [udev_hotplug_event] ignoring udev action change
^C
`

XelXen, 2022-06-14:
I read about the driver support for 04f3:0c4b on thier gitlab but found out that it is not supported yet

listout, 2022-06-14:
@XelXen It's most probably never will be supported (requires a lot of work).

XelXen, 2022-06-14:
@listout elan drivers are closed Source I guess which is why they need reverse engineering to make an open source driver for it

listout, 2022-06-14:
Yeah. The people who were successful in reverse engineering  used to
sniff packets using wireshark and then decrypt those packets. Although
I'm not an expert, I have just seen people do it that way.
If I knew how to do it, maybe I could give it a try since I've some
free time at hand now.

UnSstrennen, 2022-06-14:
@XelXen @listout but it works on Windows devices!

listout, 2022-06-14:
@UnSstrennen I know, I'm sure we all here know that. But that driver is closed source. libprintf have very recently added this device to their supported devices list. It's working, but not properly. So I guess we are heading in the right direction.

Look at this merge request https://gitlab.freedesktop.org/libfprint/libfprint/-/merge_requests/356 and https://fprint.freedesktop.org/supported-devices.html (just ctrl+f and 04f3:0c4b). I was trying out the new release and there are some issues.

XelXen, 2022-06-16:
@UnSstrennen the drivers used on windows are official Elan FP Drivers. Libfprint team is just reverse-engineering the workings of the closed-source driver to port it to Linux and sadly our device hasn't been ported yet

XelXen, 2022-06-16:
@listout Looks like u found a miracle, maybe we can reinstall libfprint and check if we got the support

listout, 2022-06-16:
@XelXen I've tested the latest release, it's not good. The guy who made the change just made sure that it works and thats about it. If you look at the gitlab thread, you can see that other people have tested it voiced their opinion. My experience is same as  those people. Verification still doesn't work.

Don't know why the developers though it was good enough to merge into a release :/

XelXen, 2022-06-16:
@listout what were the issues you faced?

listout, 2022-06-16:
- The enrollment is not perfect, does not work every time.
- I could not get verification to work, can't verify once enrolled.
- The image captured was not perfect.

XelXen, 2022-06-16:
@listout We can just wait till the patches come out

listout, 2022-06-16:
Yeah, but things are headed in right direction

XelXen, 2022-06-16:
@listout if that's the case then I'm gonna wait for devs

UnSstrennen, 2022-10-11:
https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/285

va1m, 2023-01-23:
https://gitlab.freedesktop.org/libfprint/libfprint/-/issues/285#note_1734717

XelXen, 2024-04-19:
Just checking in, the driver has finally been added and it works for the initial part but enroll process doesn't even work so I can't test it out any further.

Btw, my device is Thinkbook 15IML

jackfido, 2024-07-08:
Idk if some of you has still the issue, but to me help me this link https://support.lenovo.com/mx/es/downloads/ds560939-elan-fingerprint-driver-for-ubuntu-2204-thinkpad-e14-gen-4-e15-gen-4 since Feb 2023 Lenovo has a driver, and I use it very well on Ubuntu Unity 22.04 LTS

listout, 2024-07-09:
Does this work for E14 Gen 3?

jackfido, 2024-10-23:
> Idk if some of you has still the issue, but to me help me this link https://support.lenovo.com/mx/es/downloads/ds560939-elan-fingerprint-driver-for-ubuntu-2204-thinkpad-e14-gen-4-e15-gen-4 since Feb 2023 Lenovo has a driver, and I use it very well on Ubuntu Unity 22.04 LTS

Only to update... I use this driver in Ubuntu Unity 24.04 it work well

mppace, 2024-11-12:
Hi the above lenovo driver worked for me on Fedora 41, (on a Samsung device).  Not out of the box but was pretty simple once i put debugging on and followed the errors. Long story short, i symlinked the driver

ln -f /usr/lib64/libfprint-2-tod1-elan.so /usr/lib64/libfprint-2-tod.so.1

giacomoferretti, 2024-11-19:
It works on my Lenovo Thinkbook 16p Gen 2. I'm using Fedora 41, freshly installed.

Steps:

1. Download and extract the driver from here: https://support.lenovo.com/us/en/downloads/ds560939-elan-fingerprint-driver-for-ubuntu-2204-thinkpad-e14-gen-4-e15-gen-4
2. Install necessary packages to compile `libfprint`

	```
	sudo dnf install meson gcc gcc-c++ glib2-devel libgusb-devel gobject-introspection-devel nss-devel libgudev-devel gtk-doc
	```

3. Clone the repository and checkout to latest tod tag

	```
	git clone https://gitlab.freedesktop.org/3v1n0/libfprint.git && cd libfprint && git checkout tags/v1.94.8+tod1
	```

4. Build and install
	
	```
	meson builddir && cd builddir && meson compile && sudo meson install`
	```

5. Create links

	```
	sudo ln -sf /usr/local/lib64/libfprint-2.so.2.0.0 /usr/lib64/libfprint-2.so.2.0.0
	sudo ln -sf /usr/local/lib64/libfprint-2-tod.so.1 /usr/lib64/libfprint-2-tod.so.1
	```

6. Copy Lenovo driver

	```
	sudo mv r1slf01w/libfprint-2-tod1-elan_0.0.8_Ubuntu22.04/libfprint-2-tod1-elan.so /usr/local/lib64/libfprint-2/tod-1
	```

7. Restart `fprintd`

	```
	sudo systemctl restart fprintd
	```

8. Check for errors (optional)

	```
	journalctl -e -u fprintd.service
	```

foegra, 2025-03-26:
> Hi, the above Lenovo driver worked for me on Fedora 41, (on a Samsung device). Not out of the box but was pretty simple once i put debugging on (https://gitlab.freedesktop.org/libfprint/fprintd/-/blob/master/README) and followed the errors. Long story short,
> 
> I installed the standard fprintd packages
> 
> ```
>             sudo dnf install fprintd fprintd-pam
> ```
> 
> i symlinked the driver
> 
> ```
>            sudo ln -f /usr/lib64/libfprint-2-tod1-elan.so /usr/lib64/libfprint-2-tod.so.1
> ```
> 
> Then i had to handball this directory path for the user i was enrolling
> 
> ```
>           sudo mkdir -p /var/lib/fprint/$USER/elan/0
> ```
> 
> Restart fprintd
> 
> ```
>           systemctl restart fprintd
> ```
> 
> The you can enroll and list as normal with fprintd-enroll fprintd-list . I hope this helps.

This did not work on Thinkpad E14 G3 unfortunately. Fedora 41

foegra, 2025-03-26:
has anyone tested this?

https://forum.zorin.com/t/how-to-install-driver-for-elan-fingerprint-04f3-0c4b/32778

foegra, 2025-03-27:
> Does this work for E14 Gen 3?

this solution worked for me on Ubuntu 24.04
https://github.com/iafilatov/libfprint/issues/39#issuecomment-2430532411

foegra, 2025-04-26:
> Hi, the above Lenovo driver worked for me on Fedora 41, (on a Samsung device). Not out of the box but was pretty simple once i put debugging on [...]

On Fedora 42, ThinkPad E14 Gen 3 - this allowed to enroll finger. But every attempt to login with fingerprint - fails

isengu, 2025-05-01:
> It works on my **Lenovo Thinkbook 16p Gen 2**. I'm using **Fedora 41**, freshly installed.
> ## Steps
> 
>     1. Download and extract the driver from here: https://support.lenovo.com/us/en/downloads/ds560939-elan-fingerprint-driver-for-ubuntu-2204-thinkpad-e14-gen-4-e15-gen-4
> 
>     2. Install necessary packages to compile `libfprint`
>        ```
>        sudo dnf install meson gcc gcc-c++ glib2-devel libgusb-devel gobject-introspection-devel nss-devel libgudev-devel gtk-doc
>        ```
> 
>     3. Clone the repository and checkout to latest tod tag
>        ```
>        git clone https://gitlab.freedesktop.org/3v1n0/libfprint.git && cd libfprint && git checkout tags/v1.94.9+tod1
>        ```
> 
>     4. Build and install
>        ```
>        meson builddir && cd builddir && meson compile && sudo meson install
>        ```
> 
>     5. Create links
>        ```
>        sudo ln -sf /usr/local/lib64/libfprint-2.so.2.0.0 /usr/lib64/libfprint-2.so.2.0.0
>        sudo ln -sf /usr/local/lib64/libfprint-2-tod.so.1 /usr/lib64/libfprint-2-tod.so.1
>        ```
> 
>     6. Copy Lenovo driver
>        ```
>        sudo mv r1slf01w/libfprint-2-tod1-elan_0.0.8_Ubuntu22.04/libfprint-2-tod1-elan.so /usr/local/lib64/libfprint-2/tod-1
>        ```
> 
>     7. Restart `fprintd`
>        ```
>        sudo systemctl restart fprintd
>        ```
> 
>     8. Check for errors (optional)
>        ```
>        journalctl -e -u fprintd.service
>        ```
> 
> 
> ## Remember to redo the steps if you update `libfprint` using `dnf update`!
> 
>     1. Pull updates
>        ```
>        cd libfprint && (git pull; git checkout v1.94.9+tod1) # Update to newest tag
>        ```
> 
>     2. Build and install
>        ```
>        meson builddir --reconfigure && cd builddir && meson compile && sudo meson install
>        ```
> 
>     3. Redo the steps above from 5 to 7

My device is Thinkpad E14 gen5 and i've managed to get the fingerprint reader working in Fedora 42, however there was an issue — the issue was caused by SELinux denying access to the fingerprint module .so file. `sudo systemctl status fprintd` was showing this error:
```
Impossible to load module /usr/local/lib64/libfprint-2/tod-1/libfprint-2-tod1-elan.so: cannot open shared object file: Permission denied
```
to test if the SELinux is the issue i've executed `sudo setenforce 0` and `sudo systemctl restart fprintd` and it got working.

However the whole process seems a bit labor intensive so here is the packed version for quick and easy use in fedora 42: [https://copr.fedorainfracloud.org/coprs/dolmushcu/libfprint-tod/](https://copr.fedorainfracloud.org/coprs/dolmushcu/libfprint-tod/)

LibMeow2025, 2025-11-25:
Im having issues on **Thinkpad E15 Gen 4 (type 21ED 21EE)**.

There exists an official driver on Lenovo's website, but doesn't seem to work:
> FPC_driver_linux_27.26.23.39

Fingertip hardware ID:
```
lsusb | grep -i fingerprint
Bus 003 Device 002: ID 04f3:0c4b Elan Microelectronics Corp. ELAN:Fingerprint
```

Error log:
```
sudo systemctl status fprintd.service
○ fprintd.service - Fingerprint Authentication Daemon
Loaded: loaded (/usr/lib/systemd/system/fprintd.service; static)
Drop-In: /usr/lib/systemd/system/service.d
└─10-timeout-abort.conf
Active: inactive (dead)
Docs: man:fprintd(1) Oct 30 17:32:27 fedora fprintd[3959]: Device reported an error during identify for enroll: The driver encountered a protocol error> 
Oct 30 17:32:32 fedora fprintd[3959]: Authorization denied to :1.98 to call method 'Claim' for device 'ElanTech Fingerprint Sensor> 
Oct 30 17:33:26 fedora systemd[1]: fprintd.service: Deactivated successfully.
Oct 30 17:33:42 fedora systemd[1]: Starting fprintd.service - Fingerprint Authentication Daemon...
Oct 30 17:33:42 fedora systemd[1]: Started fprintd.service - Fingerprint Authentication Daemon.
Oct 30 17:36:02 fedora systemd[1]: fprintd.service: Deactivated successfully.
Oct 30 17:36:02 fedora systemd[1]: fprintd.service: Consumed 4.128s CPU time, 5.2M memory peak.
Oct 30 17:41:55 fedora systemd[1]: Starting fprintd.service - Fingerprint Authentication Daemon...
Oct 30 17:41:55 fedora systemd[1]: Started fprintd.service - Fingerprint Authentication Daemon.
Oct 30 17:42:25 fedora systemd[1]: fprintd.service: Deactivated successfully.
```
I can enroll my fingertip but it never passes verification:
```
fprintd-verify
Using device /net/reactivated/Fprint/Device/0
Listing enrolled fingers:
 - #0: right-index-finger
Verify started!
Verifying: right-index-finger
Verify result: verify-no-match (done)
```

I also tried running fwupdmgr, but it didn't help either.

Scydact, 2026-09-28:
Also have 04f3:0c4b working fully on a Lenovo Legion 7 (enroll, verify, sudo and lock-screen login) with the proprietary TOD driver (thanks @quanttrinh for your repackage!)

PC: Lenovo Legion 7 16IAX7 (82TD), BIOS K1CN48WW, sensor 04f3:0c4b bcdDevice 30.00 (in the power button)

OS: Fedora 44 KDE, kernel 7.2

Device: ElanTech Fingerprint Sensor, bcdDevice 30.00

Setup: fprintd 1.94.5, libfprint-tod 1.95.1, libfprint-2-tod1-elan-0c4b 0.1.0

For anyone reading, the Copr from @quanttrinh now [packages the ELAN driver too](https://github.com/iafilatov/libfprint/issues/39#issuecomment-4284788430), so there's no need to [copy binaries by hand](https://github.com/iafilatov/libfprint/issues/39#issuecomment-3577916171):
```
sudo dnf copr enable quantt/libfprint-tod
sudo dnf swap libfprint libfprint-tod
sudo dnf install libfprint-2-tod1-elan-0c4b
sudo authselect enable-feature with-fingerprint
fprintd-enroll
```

@mdjahidhasansaikat: I got it working with Lenovo's proprietary TOD driver, not mainline libfprint's open-source driver. "ElanTech Fingerprint Sensor" is the open-source driver's name; with the Lenovo driver it shows up as "ELAN Fingerprint Sensor". libfprint-2-tod1 only lets libfprint load vendor drivers. The ELAN driver itself is a separate package (libfprint-2-tod1-elan, from the ~libfprint-tod1-group PPA), so it may be worth installing that. Note that my sensor is bcdDevice 30.00 and yours is 27.01, so I can't promise it supports your firmware.

Side note for anyone whose 0c4b sits in the power button: on Windows the button ring flashes green during Windows Hello, and the Linux driver never does that. The LED turned out to be driven by the sensor itself (vendor USB register writes), so I [made a small LD_PRELOAD shim](https://github.com/Scydact/legion-elan-fp-led#installing-the-fingerprint-driver-fedora) for fprintd that sends the same commands (green while waiting, white flash on a failed match)

PedroBuffon, 2026-10-10:
Can confirm @Scydact method works on my thinkpad E14 gen 5 (AMD), the script for making it flash green works too, it made the little LED flash green

jackfido, 2026-10-10:
Hello Again people!

Im here just to share my experience making work this driver into Debian 13 Trixie in a **ThinkBook 14 G2 ITL**, If somebody, as me, are using debian, you can keep this steps to make it work

Debian 13 Trixie (LED doent work)

Debian has an open source driver but it fails, then u need to compile the TOD propietary driver, (I did this with IA help)

1. Install dependencies
```

   sudo apt update
   sudo apt install meson ninja-build libgusb-dev libgudev-1.0-dev libnss3-dev libpixman-1-dev libgirepository1.0-dev gtk-doc-tools libpam0g-dev libudev-dev build-essential git
```

In my specific case debian 13 trixie was looking driver in the open source path **udev.pc** file (idk why but that told me the IA) then I must to create a simlink from **libudev.pc** to that path: 

   ```sudo ln -sf /usr/lib/x86_64-linux-gnu/pkgconfig/libudev.pc /usr/lib/x86_64-linux-gnu/pkgconfig/udev.pc```

2. Compile libfprint with TOD support:

```
git clone https://gitlab.freedesktop.org/3v1n0/libfprint.git
cd libfprint
meson setup build -Dtod=true -Dudev_rules=disabled
ninja -C build
sudo ninja -C build install
sudo ldconfig
cd ..
```

3. Download the propietary driver (download the Ubuntu 22.04 driver)
https://support.lenovo.com/us/en/downloads/ds560939-elan-fingerprint-driver-for-ubuntu-2204-thinkpad-e14-gen-4-e15-gen-4?spm=a2ty_o01.29997173.0.0.575ac921RLTlxh

4. And copy to this path:
```
 sudo mkdir -p /usr/local/lib/x86_64-linux-gnu/libfprint-2/tod-1/
   sudo cp libfprint-2-tod1-elan.so /usr/local/lib/x86_64-linux-gnu/libfprint-2/tod-1/
```

5. Configure driver to use TOD:

```
sudo mkdir -p /etc/systemd/system/fprintd.service.d/
   echo -e "[Service]\nEnvironment=\"FP_TOD_DRIVERS_DIR=/usr/local/lib/x86_64-linux-gnu/libfprint-2/tod-1\"" | sudo tee /etc/systemd/system/fprintd.service.d/override.conf
```

6. Reload service:

   ```
sudo systemctl daemon-reload
   sudo systemctl restart fprintd.service
```

7. Create a rule:
 ```sudo nano /etc/udev/rules.d/60-libfprint-2-tod1-elan.rules```

whit this content:

```
   SUBSYSTEM=="usb", ATTRS{idVendor}=="04f3", ATTRS{idProduct}=="0c4b", ATTRS{dev}=="*", TEST=="power/control", ATTR{power/control}="on", MODE="0660", GROUP="plugdev"
   SUBSYSTEM=="usb", ATTRS{idVendor}=="04f3", ATTRS{idProduct}=="0c4b", ENV{LIBFPRINT_DRIVER}="Elan Fingerprint Sensor"
```

Apply the rule:
```
   sudo udevadm control --reload-rules
   sudo udevadm trigger --subsystem-match=usb
   sudo systemctl restart fprintd.service
```

And finally verify:
   ```fprintd-list $USER```

If you see "ELAN Fingerprint Sensor" intead of **ElanTech Fingerprint Sensor** u got it

$ fprintd-list $USER
found 1 devices
Device at /net/reactivated/Fprint/Device/0
Using device /net/reactivated/Fprint/Device/0
Fingerprints for user user on **ELAN Fingerprint Sensor** (press):
 - #0: right-index-finger

And then u can enroll it