iafilatov/libfprint #42: 04f3:0c00 Elan Microelectronics Corp. ELAN:ARM-M4

ArtemPavliukovich, 2022-02-09:
This device isn't available, how I can fix it?

All3xJ, 2023-01-09:
I installed the enalmoc2 version in Fedora 37 (I have 04f3:0c00 fingerprint reader).
I did:

```
git clone -b elanmoc2 https://gitlab.freedesktop.org/Depau/libfprint/
cd libfprint/
sudo dnf intall glib2-devel libgusb-devel gobject-introspection-devel pixman-devel nss-devel libgudev-devel gtk-doc
meson setup builddir && cd builddir
sudo ninja install
fprintd-enroll
```

 but I get: `Impossible to enroll: GDBus.Error:net.reactivated.Fprint.Error.NoSuchDevice: No devices available` 

what am I doing wrong?

All3xJ, 2023-01-15:
I FIXED! I did what this comment says: https://gitlab.freedesktop.org/libfprint/libfprint/-/merge_requests/330#note_1718869 and then rebuilded and reinstalled as usual and restarted the service with `sudo systemctl restart fprintd.service` and now it works!

ZeroKun265, 2023-03-24:
@All3xJ how did you get it to work? Did you clone the repo from the PR that had the patch and built manually?
I'm not that well versed in this these things and i just want my fingerprint reader (04f3:0c00) to work haha

mh-saeed, 2024-02-01:
I'm using Ubuntu 23.10 &I have the same fingerprint device on my laptop.
How can I make it work?

mh-saeed, 2024-02-03:
Fixed  :heavy_check_mark: 

I just followed the instructions from this comment: https://gitlab.freedesktop.org/libfprint/libfprint/-/merge_requests/330#note_1718869

sagarbangade, 2024-06-25:
@All3xJ bro I am new to linux and all could you please explain me how did you fixed it through https://gitlab.freedesktop.org/libfprint/libfprint/-/merge_requests/330#note_1718869    I am getting confused :.(