iafilatov/libfprint #47: fprint enroll not keeping fingerpints

mppace, 2024-06-29:
Fedora FC40 with 04f3:0c6e Elan Microelectronics Corp. ELAN:Fingerprint

So using the Gnome fingerprint reader error when it starts to scan fingerprint with device claimed by another process.  You can kill the process and start again, but it does the same.

fprintd-enrol seems to work

**_root_root@fedora:/usr/lib64# fprintd-enroll
Using device /net/reactivated/Fprint/Device/0
Enrolling right-index-finger finger.
Enroll result: enroll-stage-passed
Enroll result: enroll-disconnected__**

but list shows empty

found 1 devices

**Device at /net/reactivated/Fprint/Device/0
Using device /net/reactivated/Fprint/Device/0
User root has no fingers enrolled for ElanTech Fingerprint Sensor.**

Syslog shows

**Jun 29 09:18:10 fedora fprintd[11792]: Device reported an error during enroll: The driver encountered a protocol error with the device.**

This kill the systemd process and needs to be restarted.

I've tried the fedora supplied fprintd v1.94.2 but also manually installed v.1.94.6 and v.1.94.7

If there is any debugging you want me to do please let me know how I can help.

Kind regards

Paul

micahlt, 2025-07-17:
I encountered the same issue and couldn't solve it.

vrtgs, 2026-06-29:
same issue on `Bus 003 Device 003: ID 04f3:0c4f Elan Microelectronics Corp. ELAN:Fingerprint`